CobraPi/Klipper-Firmware-FLSUN-QQ-S-Pro-Delta-Printer #3: Baud unstable

BugsonPL, 2021-05-19:
I find the default baud unstable with Flsun highspeed 1.0 board.
115200 works for me fine.

If you could confirm it is worth changing.

CobraPi, 2021-05-20:
Hmm that's strange. The default baud rate of 250000 works pretty well on my Highspeed board. Others seem to have no issue with it. It's even recommended in the documentation: 

https://www.klipper3d.org/FAQ.html#how-do-i-change-the-serial-baud-rate

closed 2021-06-01 by CobraPi

BugsonPL, 2021-06-02:
With other boards I also use 250000 but with this one it is not stable. I don't see any problems with performance.

CobraPi, 2021-06-17:
Okay, since only a minority of people have this problem I'll add it to the troubleshooting section of the README.