CobraPi/Klipper-Firmware-FLSUN-QQ-S-Pro-Delta-Printer #8: Issues running your configuration on FLSun QQ-S Pro

mifrendjim, 2021-11-07:
I followed your guide, and so far I'm having issues with the steppers A, B, C. 

I'm getting a ton of vibration from the stock motors.
Only 2 of the 3 arms are homing correctly, but hitting the end stops.
When I try to test extrude Klipper looks like it's crashing, or the firmware is crashing.
Klipper is also not allowing me to move the gantry around using the controls. It sends the command but returns 
`Move out of range: 0.000 10.000 370.010 [0.000]`

I had to update the printer.cfg so all the stepper blocks had:
`position_endstop: 370
 arm_length: 280.0`

Did you run into any of these issues? I've attached my config.
[printer.txt](https://github.com/CobraPi/Klipper-Firmware-FLSUN-QQ-S-Pro-Delta-Printer/files/7493138/printer.txt)

conlank, 2021-11-08:
Post the klipper.log

mifrendjim, 2021-11-10:
`===== Config file =====
[mcu]
serial = /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0
restart_method = command
baud = 250000

[printer]
kinematics = delta
max_velocity = 500
max_accel = 2000
max_z_velocity = 200
minimum_z_position = -5
delta_radius = 130

[stepper_a]
step_pin = PE3
dir_pin = PE2
enable_pin = !PE4
rotation_distance = 32
endstop_pin = PA15
microsteps = 16
homing_speed = 20
position_endstop = 370
arm_length = 280.0

[stepper_b]
step_pin = PE0
dir_pin = PB9
enable_pin = !PE1
rotation_distance = 32
endstop_pin = PA12
microsteps = 16
homing_speed = 20
position_endstop = 370
arm_length = 280.0

[stepper_c]
step_pin = PB5
dir_pin = PB4
enable_pin = !PB8
rotation_distance = 32
endstop_pin = PC4
microsteps = 16
homing_speed = 20
position_endstop = 370
arm_length = 280.0

[probe]
pin = !PA11
x_offset = 0
y_offset = 0
z_offset = 10
samples = 3
samples_result = average
sample_retract_dist = 5

[delta_calibrate]
radius = 120
horizontal_move_z = 10

[extruder]
step_pin = PD6
dir_pin = !PD3
enable_pin = !PB3
rotation_distance = 9
microsteps = 16
nozzle_diameter = 0.400
filament_diameter = 1.750
heater_pin = PC3
sensor_type = EPCOS 100K B57560G104F
[... 414 lines cut ...]
# CONFIG_STM32_CANBUS_PB8_PB9 is not set
CONFIG_CANBUS_FREQUENCY=500000
CONFIG_SERIAL_BAUD=250000
CONFIG_USB_VENDOR_ID=0x1d50
CONFIG_USB_DEVICE_ID=0x614e
CONFIG_USB_SERIAL_NUMBER_CHIPID=y
CONFIG_USB_SERIAL_NUMBER="12345"
CONFIG_INITIAL_PINS=""
CONFIG_HAVE_GPIO=y
CONFIG_HAVE_GPIO_ADC=y
CONFIG_HAVE_GPIO_SPI=y
CONFIG_HAVE_GPIO_I2C=y
CONFIG_HAVE_GPIO_HARD_PWM=y
CONFIG_HAVE_GPIO_BITBANGING=y
CONFIG_HAVE_STRICT_TIMING=y
CONFIG_HAVE_CHIPID=y
CONFIG_HAVE_STEPPER_BOTH_EDGE=y
CONFIG_INLINE_STEPPER_HACK=y
=======================
Build file /home/pi/klipper/klippy/../out/klipper.dict(6839): Sun Nov  7 20:49:03 2021
Last MCU build version: v0.10.0-113-gd4aee4f5
Last MCU build tools: gcc: (15:7-2018-q2-6) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907] binutils: (2.31.1-11+rpi1+11) 2.31.1
Last MCU build config: BUS_PINS_i2c1=PB6,PB7 RECEIVE_WINDOW=192 RESERVE_PINS_serial=PB11,PB10 BUS_PINS_i2c2=PB10,PB11 BUS_PINS_i2c1a=PB8,PB9 STEPPER_BOTH_EDGE=1 SERIAL_BAUD=250000 ADC_MAX=4095 BUS_PINS_spi3=PB4,PB5,PB3 BUS_PINS_spi2=PB14,PB15,PB13 BUS_PINS_spi1=PA6,PA7,PA5 PWM_MAX=255 MCU=stm32f103xe STATS_SUMSQ_BASE=256 BUS_PINS_spi1a=PB4,PB5,PB3 CLOCK_FREQ=72000000
Build file /home/pi/klipper/klippy/../out/klipper.elf(251836): Sun Nov  7 20:49:17 2021
`

conlank, 2021-11-10:
From your log it looks like you might need to redo the steps in [here](https://github.com/CobraPi/Klipper-Firmware-FLSUN-QQ-S-Pro-Delta-Printer#step-3----connecting-the-printer-to-the-raspberry-pi). Specifically the one's about setting up the serial ID. The move out of range at the top makes sense because the delta is allowed to move in a shape that is toped with a cone, so the highest point (Z=370) means you are stuck at X,Y of 0,0. Try moving lower before moving in the X and Y.

CobraPi, 2021-11-13:
I also noticed that you left some parameters suck as the position endstops, arm length, and homing speed (not that important). However, if your delta arms are set at an arbitrary value, that could cause uneven homing of each axis. 

According to your log, it seems like your microcontroller is having trouble establishing a serial connection to the Pi. 

I'd look into possibly try to reflash it agaihn

chilledflo, 2021-11-18:
Unfortunatly Im having the same issue on my qq-s pro. Any Solutions came up?

conlank, 2021-11-23:
To me, it looks like the arms and end-stops are set to the default lengths. This can cause issues with movement not being true to reality. Also, extreme movements like @mifrendjim showed in his error (X0, Y10, Z370) _are_ outside of the movement area, so this error is not an error that is possible to fix. 

My recommendation is to run probe_calibrate and then delta_calibrate. Make sure your probe is attached and plugged in.