CobraPi/Klipper-Firmware-FLSUN-QQ-S-Pro-Delta-Printer #11: No com port and errors in the .cfg

pingywon, 2022-01-12:
I have a QQS Pro. 
I am using your .cfg for my printer.

1) no matter what I do I can not get OctoKlipper to connect. 
2) When I do try I get the errors about:
`// Option 'default_parameter_x' is not valid in section 'gcode_macro pause'
//
// Once the underlying issue is corrected, use the "RESTART"
// command to reload the config and restart the host software.
N0 M110 N0*125
// Printer is halted
!! Option 'default_parameter_x' is not valid in section 'gcode_macro pause'`

I am admittedly over my head. I was hoping to some easy to follow directions for octo and klipper on the FLSUN QQ-S Pro. Everything seemed to be working until I got to this point.

ANY help would be greatly appreciated.

pingywon, 2022-01-12:
`# Pause print macro
[gcode_macro PAUSE]
rename_existing: BASE_PAUSE
default_parameter_X: 230    #edit to your park position
default_parameter_Y: 230    #edit to your park position
default_parameter_Z: 10     #edit to your park position
default_parameter_E: 5      #edit to your retract length
gcode:
    SAVE_GCODE_STATE NAME=PAUSE_state
    BASE_PAUSE
    G91
    G1 E-{E} F2100
    G1 Z{Z}
    G90
    G1 X{X} Y{Y} F6000

# Resume print macro
[gcode_macro RESUME]
rename_existing: BASE_RESUME
default_parameter_E: 5      #edit to your retract length
gcode:
    G91
    G1 E{E} F2100
    G90
    RESTORE_GCODE_STATE NAME=PAUSE_state MOVE=1
    BASE_RESUME`

`default_parameter_X: 230    #edit to your park position
default_parameter_Y: 230    #edit to your park position
default_parameter_Z: 10     #edit to your park position
default_parameter_E: 5      #edit to your retract length`

These default params see to be removed from Klipper. Can I just comment them out? I have seen one example that was way over my head. If anyone can give me good starting vaules or replacments for this and:
`# Resume print macro
[gcode_macro RESUME]
rename_existing: BASE_RESUME
default_parameter_E: 5      #edit to your retract length`

pingywon, 2022-01-12:
If I reboot everything I can get connected. Then I run the G28 with out issues. Once I run the PROBE_CALIBRATION the printer/klipper connection times out before the probe reaches the build plate.

pingywon, 2022-01-13:
Thanks to [This pull](https://github.com/CobraPi/Klipper-Firmware-FLSUN-QQ-S-Pro-Delta-Printer/pull/12)

I was able to get my macros cleared up, but I am still not able to get a connection to the printer. If I shut everything down. unplug the USB and then start it all up I can get a connection to the printer through Octo SOMETIMES, but as soon as I try to `PROBE_CALIBRATE` the connection times out and I can not get reconnected unless I follow the above steps.

pingywon, 2022-01-13:
Once the COM gets broken it never reconnects with out and bunch of reboots of both the pi and the printer (still havent figured out which in what order makes it work again) now I am getting this error

`// Can not update MCU 'mcu' config as it is shutdown
// Once the underlying issue is corrected, use the
// "FIRMWARE_RESTART" command to reset the firmware, reload the
// config, and restart the host software.
// Error configuring printer
!! Can not update MCU 'mcu' config as it is shutdown`

conlank, 2022-01-13:
I've never used octoklipper, but it sounds like you might need to make sure the `serial:` parameter in your printer.cfg is correct. [The steps in this section](https://github.com/CobraPi/Klipper-Firmware-FLSUN-QQ-S-Pro-Delta-Printer#step-3----connecting-the-printer-to-the-raspberry-pi) might provide some insight, but as I have not used octoklipper I don't think I will be much help.

pingywon, 2022-01-14:
> I've never used octoklipper, but it sounds like you might need to make sure the `serial:` parameter in your printer.cfg is correct. [The steps in this [...]

Hey I understand and thank you for the reply. I have indeed read that section (over and over) Im not doing it wrong because sometimes I can get it to connect but then it is real touchy one false move and it times out and I cant reconnect with out rebooting the printer and my RPi. 
I have dropped back to stock firmware, I guess me and klipper just werent meant to be.

`pi@octopi:~ $ ls /dev/serial/by-id/*
/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0` ....and all that

conlank, 2022-01-14:
> Hey I understand and thank you for the reply. I have indeed read that section (over and over) Im not doing it wrong because sometimes I can get it to connect but then it is real touchy one false move and it times out and I cant reconnect with out rebooting the printer and my RPi.

I can't tell if the issue is with Octoprint talking to Klipper, or Klipper talking to the printer. If it's Octoprint related then the OctoKlipper github issues page might be more helpful. Can you try setting up with Mainsail or Fluidd to rule out if the issue is on the Octoprint end?

pingywon, 2022-01-17:
> I can't tell if the issue is with Octoprint talking to Klipper, or Klipper talking to the printer. If it's Octoprint related then the OctoKlipper github issues page might be more helpful. Can you [...]

I can load another SD card with mainsail to try and get to the bottom of it. I will say I have been running octoprint on this printer for over a year and never had an serial com issues. So I have no idea what is going on.

pingywon, 2022-01-20:
> I can't tell if the issue is with Octoprint talking to Klipper, or Klipper talking to the printer. If it's Octoprint related then the OctoKlipper github issues page might be more helpful. Can you [...]

So I loaded Mainsail OS and got everything set up. Im still having an issue. After adding my printer.cfg to mainsail and trying to get it to connect I get this error:

mcu 'mcu': Command format mismatch: config is_config=%c crc=%u move_count=%hu is_shutdown=%c vs config is_config=%c crc=%u is_shutdown=%c move_count=%hu

This type of error is frequently caused by running an older
version of the firmware on the micro-controller (fix by
recompiling and flashing the firmware).

Known versions: host=v0.9.1-743-g95047784, mcu=v0.10.0-164-ga1af78cb

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Protocol error connecting to printer

conlank, 2022-01-21:
You need to update klipper on the raspberry pi. ```Known versions: host=v0.9.1-743-g95047784, mcu=v0.10.0-164-ga1af78cb```

pingywon, 2022-01-24:
Thanks man. I ran apt update-upgrade and didnt get anything. Either way Im taking a break I thank you but I have been nothing but frustrated and I need to get some things printed. I switched back to FLSUN firmware.

closed 2022-01-24 by pingywon