CobraPi/Klipper-Firmware-FLSUN-QQ-S-Pro-Delta-Printer #13: Option 'uart_pin' in section 'tmc2209 stepper_a' must be specified

Xcat2008, 2022-01-20:
Hello, thank you in advance. 
I'm stuck on this error message :/ I have 2208 drivers, dunno if is that the main problem on my QQS-Pro. If i apply your printer config Stock, everything works, less the HOME ALL but maybe is backwards. When applying your config with 2209 got that error... I do not know what do more.. Already spend a lot of hours truing to figure out, i know from the message i need to set a number or something for UART connection, but dunno what set there. Can you help? Got also one video, because my english is not that great. Thanks https://youtu.be/vzhqECSI52o 
[image]
[image]

conlank, 2022-01-21:
In the TMC 2208/9 Config the UART pins are not defined due to the variety of ways they could be attached. You will need to verify which pins are connected to which steppers and assign the UART pins accordingly. This [page](https://github.com/Foxies-CSTL/Marlin_2.0.x/wiki/3.SPECIAL-MODS#31-TMC-with-UART-mode) should contain the information you need to correctly specify the uart pins.

CobraPi, 2022-01-30:
Yes, you need to define a [TMC 2208/9] section in your .cfg file if you're going to use UART. I've never done this UART mod to my QQ-S though.

Xcat2008, 2022-01-30:
Thank you for the answer.. Ive figure out i had to take 1 switch in each driver, and uncoment almost the last configs for the Pin's 

> #######################################################################
# Fairly quiet settings without running stealthchop if you have TMC2209 Drivers. Likely works with 2208 but has not been tested.
#######################################################################

[tmc2209 stepper_a]
#uart_pin: PA10 
#These pins are commented so you have to check which pins are connected to which steppers.
run_current: .850
hold_current: .500
interpolate: True
stealthchop_threshold: 0

[tmc2209 stepper_b]
#uart_pin: PA9
run_current: .850
hold_current: .500
interpolate: True
stealthchop_threshold: 0

[tmc2209 stepper_c]
#uart_pin: PC7
run_current: .850
hold_current: .500
interpolate: True
stealthchop_threshold: 0

[tmc2209 extruder]
#uart_pin: PA8
interpolate: True
run_current: .850
hold_current: .500
stealthchop_threshold: 0 <

The problem was, when Home ALL now, did not work properly and then the Shutdown message appears... 
But now does not matter, maybe noobish from me, never use this.. i start messing around taking all switch's from the drivers, dunno what happends the MCU burn out.. Board is dead... I'm waiting for a new mobo, mks robin nano v3.1

conlank, 2022-05-04:
Please make sure you have the driver jumper wires plugged into the pins specified on this page. https://github.com/Foxies-CSTL/Marlin_2.0.x/wiki/3.SPECIAL-MODS#31-TMC-with-UART-mode