CobraPi/Klipper-Firmware-FLSUN-QQ-S-Pro-Delta-Printer #14: Vref pin configuration missing from the qq-s stock configuration file

shamilabw, 2022-02-02:
This is the configs I had to add,

[output_pin vref_xy]
pin: PA6
pwm: True
shutdown_value: 0
scale: 3.3
cycle_time: .0001
value: 1.6

[output_pin vref_z]
pin: PA7
pwm: True
shutdown_value: 0
scale: 3.3
cycle_time: .0001
value: 1.6

[output_pin vref_e]
pin: PB0
pwm: True
shutdown_value: 0
scale: 3.3
cycle_time: .0001
value: 1.6

CobraPi, 2022-03-17:
This is related to your specific setup. Most of people don't need to add this parameter to get Klipper to work.

closed 2022-03-17 by CobraPi

meteyou, 2022-05-04:
would be nice to have these vRef settings in the default config. without these settings, it could work, but have not to work.

I only would change the names to `_vref_xy`, `_vref_z` & `_vref_e` to hide these values in the GUI.