CobraPi/Klipper-Firmware-FLSUN-QQ-S-Pro-Delta-Printer #18: QQ-S Pro Movement Distance is Off

Mousedroid98, 2022-02-07:
Hi! I'm having issues with my QQ-S Pro install of Klipper. I had this install working just fine last week, but an unfortunate micro-sd corruption incident required me to fresh install it again. However, this new time around, I'm finding that the printer seems to think it's moving far less than it is (e.g. the bed is registered as 190-ish mm). This is wreaking havoc with the printer as I can't properly calibrate or level the printer.

I'm running the TMC2209 config with standalone (turned off the UART sections, which worked for me last time), and I've checked VRefs and such. Not sure what's causing this issue or how to fix it, but I'd like to keep using this excellent Klipper config. Thanks for the help!

conlank, 2022-02-07:
Rotation distance is based on the physical printer. Running the TMC2209 in Standalone mode will cause them to set their micro-steps using the jumpers on the board, the fact that this is corrected by changing `rotation_distance`, which should be calculated with this formula `rotation_distance = <belt_pitch> * <number_of_teeth_on_pulley>` suggests this is the case. You could change the `microsteps: 16` in the config to `microsteps: 32` and keep the rotation distance set to 32, as if your rotation distance is doubled and you enable UART, you will run into this problem again.

Mousedroid98, 2022-02-08:
Excellent, I appreciate the help! Thanks!

closed 2022-02-08 by Mousedroid98