CobraPi/Klipper-Firmware-FLSUN-QQ-S-Pro-Delta-Printer #20: QQs vs QQs pro

NunoFilipePereira, 2022-05-04:
What modifications do I have to make to work in Klipper between qqs pro and qqs? both are the same and sometimes even bring the same board. However I'm not getting Klipper on qqs with the boatd robin nano mini v2.0.

Transpupino, 2022-06-07:
[image]
You need add section OUTPUT PIN for VREF on Robin Mini V2.0.
On this board A4988 without resistors for Vref, it generates with PWM by STM32.

drowhunter, 2023-02-20:
I don't understand this,  Is this needed for QQs support still?  What file do I have to edit?

ch4d20, 2024-08-18:
# Klipper config file for QQ-S Pro with:
# - BL touch
# - Filament Sensor

# Mesh bed leveling is also added, but commented out. Only enable it if
# running the PROBE_ACCURACY command outputs a range value < 0.025 mm

# Note that an offset probe is not recommended on delta printers due to
# effector tilt. In order to use it successfully to create a useable bed mesh
# you need to run the enhanced delta calibration:
# https://www.klipper3d.org/Delta_Calibrate.html

# See docs/Config_Reference.md for a description of parameters.

[mcu]
serial: /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0
restart_method: command
baud: 250000

[printer]
kinematics: delta
max_velocity: 500
max_accel: 2000
max_z_velocity: 200
minimum_z_position: 0
delta_radius: 130

[stepper_a]
step_pin: PE3
dir_pin: PE2
enable_pin: !PE4
rotation_distance: 32
endstop_pin: PA15
microsteps: 16
homing_speed: 60
position_endstop: 370
arm_length: 280.0

[stepper_b]
step_pin: PE0
dir_pin: PB9
enable_pin: !PE1
rotation_distance: 32
endstop_pin: PA12
microsteps: 16

[stepper_c]
step_pin: PB5
dir_pin: PB4
enable_pin: !PB8
rotation_distance: 32
endstop_pin: PC4
microsteps: 16

[extruder]
step_pin: PD6
dir_pin: !PD3
enable_pin: !PB3
rotation_distance: 9
microsteps: 16
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: PC3
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PC1
control: pid
pid_Kp: 14.529
pid_Ki: 0.557
pid_Kd: 94.802
min_extrude_temp: 180
min_temp: 0
max_temp: 260

[heater_bed]
heater_pin: PA0
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PC0
control: pid
pid_Kp: 325.10
pid_Ki: 63.35
pid_Kd: 417.10
min_temp: 0
max_temp: 150

[fan]
pin: PB1
kick_start_time: 0.200

#######################################################################
# Optional: Bed mesh leveling - only use if the PROBE_ACCURACY
#                               command outputs a range value < 0.025 mm
#######################################################################

[bed_mesh]
speed: 50
horizontal_move_z: 5
mesh_radius: 100
mesh_origin: 0,0
round_probe_count: 5
algorithm: lagrange

#######################################################################
# Optional: End-stop style filament sensor
#######################################################################

[filament_switch_sensor Filament]
pause_on_runout: True
switch_pin: PA4

#######################################################################
# Optional: BL Touch Probe - comment out [probe] section if using this
#######################################################################

[bltouch]
sensor_pin: PA11
control_pin: PA8
stow_on_each_sample: True
set_output_mode: 5V
x_offset: 3
y_offset: -36
z_offset: 0.01
samples: 3
sample_retract_dist: 5
samples_result: average

#######################################################################
# Additional Output Pins for VREF
#######################################################################

[output_pin VREF_XY]
pin: PA6
pwm: True
static_value: 0.5
cycle_time: 0.001
hardware_pwm: True

[output_pin VREF_Z]
pin: PA7
pwm: True
static_value: 0.5
cycle_time: 0.001
hardware_pwm: True

[output_pin VREF_E]
pin: PB0
pwm: True
static_value: 0.38
cycle_time: 0.001
hardware_pwm: True

# This is the path where your g-code files will be uploaded to in the
# Linux PC
[virtual_sdcard]
path: /home/your-username/gcode_files

[display_status]

[pause_resume]

# This macro should be called at the 'start g-code' part
# of your slicer and you would put all the start g-code here
# instead so that you don't have to re-slice everytime you
# have to edit the commands
[gcode_macro START_PRINT]
gcode:
    #SET_GCODE_OFFSET Z_ADJUST=-0.1 MOVE=1 # sets the z-offset

# This macro should be called at the 'end g-code' part
# of your slicer and like the macro above, you would
# also put all of your end g-code here for the same reason as above
[gcode_macro END_PRINT]
gcode:
    SET_GCODE_OFFSET Z_ADJUST=0 # resets the z-offset (otherwise, )

# Pause print macro
[gcode_macro PAUSE]
rename_existing: BASE_PAUSE
gcode:
    SAVE_GCODE_STATE NAME=PAUSE_state
    BASE_PAUSE
    G91
    G1 E-{params.E|default(1)} F2100 ; change 1 to your default retraction distance
    G1 Z{params.Z|default(10)} ; change 10 to your Z parking position
    G90
    G1 X{params.X|default(230)} Y{params.Y|default(230)} F6000 ; change 230 to your X and Y parking position

# Resume print macro
[gcode_macro RESUME]
rename_existing: BASE_RESUME
gcode:
    G91
    G1 E{params.E|default(1)} F2100 ; change 1 to your default retraction distance
    G90
    RESTORE_GCODE_STATE NAME=PAUSE_state MOVE=1
    BASE_RESUME

# Cancel print macro
[gcode_macro CANCEL_PRINT]
rename_existing: BASE_CANCEL_PRINT
gcode:
    TURN_OFF_HEATERS
    CLEAR_PAUSE
    SDCARD_RESET_FILE
    BASE_CANCEL_PRINT

#*# <---------------------- SAVE_CONFIG ---------------------->
#*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
#*#
#*# [bltouch]
#*# z_offset = 0.005

ch4d20, 2024-08-18:
why isnt my bl touch working ?